SpiceSharp/SpiceSharp #2: Problem about voltage loop

CIVITAS-John, 2017-10-30:
Hi Sven, 
I've tried your spice# and find a problem which I don't know whether belongs to mine or your library.

Circuit ckt = new Circuit();
ckt.Objects.Add(
		new Voltagesource("V1", "6", "gnd", 1),
		new Resistor("V1-R", "gnd", "1", 1),
		new Voltagesource("V2", "1", "2", 1),
		new Resistor("V2-R", "2", "3", 1),
		new Voltagesource("V3", "3", "4", 1),
		new Resistor("V3-R", "4", "5", 1),
		new Resistor("R", "1", "6", 1)
	);

It shall throw an exception notifying 'voltage loop found' when performing checks.

svenboulanger, 2017-10-30:
Hello @CIVITAS-John 

First of all, thank you for trying out Spice#. Your feedback is much appreciated!
Now on with the issue. I believe this is a bug in Spice#. The method for checking voltage loops needs some rework too.

Yours sincerely,
Sven

CIVITAS-John, 2017-10-30:
Many thanks for you reply!

I also detected a potential bug, that the order of Objects.Add may influence the output of floating node checking. I don't know whether it is by design or not, if not, I shall try to construct some examples for you.

CIVITAS-John, 2017-10-30:
Circuit ckt = new Circuit();
ckt.Objects.Add(
		new Voltagesource("V2", "1", "2", 1),
		new Voltagesource("V1", "6", "gnd", 1),
		new Resistor("V1-R", "gnd", "1", 1),
		new Resistor("V2-R", "2", "3", 1),
		new Resistor("R", "1", "6", 1)
	);

This example also throws an exception of voltage loop. No problem when in normal order.

svenboulanger, 2017-10-30:
You can try using the development branch for a fix.

CIVITAS-John, 2017-10-31:
Thanks! I will test it as soon as possible. 
Btw can you add a checking mode to automatically remove floating components rather than throw an exception? Or can you make the exception contains relevant identifier in a separate property (e.g CircuitIdentifier[] NodesConcerned) so that we can remove them in-code?

svenboulanger, 2017-10-31:
I will have to think a bit about that. I know that often circuit simulators add a GMIN conductance to ground for floating nodes while showing a warning instead of raising an error. We should also take care that some circuit objects cannot be removed under that rule.

In any case, the checking-methods do need a lot of work so I am keeping this issue open. It would be nice if we could make checking circuits modular as well such that third parties can impose custom rules on their circuits.

CIVITAS-John, 2017-10-31:
Temporarily, I've taken liberty to slightly modify your Check() using code below:

// Check for floating nodes
var unconnected = FindFloatingNodes();
if (unconnected.Count > 0)
{
	List<CircuitIdentifier> FloatingNodes = new List<CircuitIdentifier>();
	for (int i = 0; i < ckt.Nodes.Count; i++) {
		int index = ckt.Nodes[i].Index;
		if (unconnected.Contains(index))
			FloatingNodes.Add(ckt.Nodes[i].Name);
	}
	return FloatingNodes;
}
return null;

I believe that a well-designed checking pipeline can solve many problems relating to it. Well done!

commit ec8da348 (2017-10-30): Fixed #2
closed 2018-03-26 by svenboulanger