SpiceSharp/SpiceSharp #4: Allowing dynamic behaviour executing during transient simulation

CIVITAS-John, 2017-11-01:
We're currently trying to use Spice# for some interactive circuit simulation, which requires to slightly modify components' parameters during simulation loop (we use event to intercept iteration and inject modification). In current release we can call temperature function, yet after switching to development branch the function has been moved; we therefore create a ResistorTemperatureBehavior and call it instead. 

I'm not so familiar with this kind of Spice usage before (since this is the first thing that can work as a library), and I don't know whether my current method will cause some unknown problems. If that's really possible (in our test cases that really works, but I am not sure about its theoretical certainty), maybe you can allow your main library do it without hijacking.

svenboulanger, 2017-11-01:
Hello @CIVITAS-John ,

As you found out, we are currently refactoring everything to accommodate adding custom components AND simulations more easily. This is still in development so it is subject to change and will be part of the 2.0 to come.

The analysis execution flow is typically:
- When starting any simulation, the circuit is set up (Setup). In this method, nodes are mapped into the circuit. This means that devices cannot change the number of nodes in the circuit anymore after the analysis has started.
- After setup, the temperature behavior is called once. Here, any parameters that depend on the temperature (`ckt.State.Temperature`) are calculated.
- After that, any other behavior used will/should depend on the analysis running, and they can use the parameters calculated in the last temperature behavior.

So it should certainly be possible to call the temperature behaviors mid-analysis and continue using these newly calculated parameters. The only issues I could think of are:
- In transient analysis, convergence might fail if the temperature change is too drastic.
- Changing temperature mid AC/Noise-analysis could cause changes in the operating point which will not be reflected in the spectrum as the OP is only called once at the start.

Kind regards,
Sven

svenboulanger, 2018-10-08:
Please refer to issue #17 as well.

I wrote documentation on this here:
[https://spicesharp.github.io/SpiceSharp/articles/changing_parameters/example.html](https://spicesharp.github.io/SpiceSharp/articles/changing_parameters/example.html).

We will keep this supported as we continue with Spice#.

closed 2018-10-08 by svenboulanger