SpiceSharp/SpiceSharp #7: AC simulation uses wrong solution

svenboulanger, 2017-11-08:
Most devices will use OldSolution or the last state variable (which are solutions to the previous iteration) as opposed to Solution (the solution of the current iteration) to calculate small-signal parameters for AC analysis. I believe this is not logical, as the most recent iteration should be a better approximation to the real solution.

Note that changing this breaks tests, because Spice 3f5, which is used as a reference, does this as well. I do not think other Spice-derivatives will have changed this (from what I could see ngSpice didn't).

svenboulanger, 2017-11-11:
One way of resolving this is simulating using Spice 3f5 with very strict tolerances. This way we get something very close to the "true" solution of the model, and the differences how Spice# solves it become less important, as long as it is correct (ie. within tolerance levels to this "true" solution).

CIVITAS-John, 2017-11-11:
Maybe you can add a setting option about it, like 'Spice 3f5 Compatible' and 'Something Greater'. lol

svenboulanger, 2017-11-13:
Adding a settings flag for some kind of legacy mode would make sense for a Spice clone, however that is not the direction I'd like Spice# to take. I'd like to improve upon it. I believe the main concern should be that the models and simulations lead to accurate results.

svenboulanger, 2017-11-22:
Using strict tolerances works for DC analysis, but still needs some work for AC, Transient and Noise analysis. Maybe it's better to test the model behaviors separately. I'll look into what's possible.

svenboulanger, 2017-11-29:
We modified Spice 3f5 a little bit such that it uses the last iteration instead.
I am convinced that it is a bug in Spice 3f5, especially after finding some code that **does** use the last iteration.
Please note that none of the simulation results will deviate significantly because that is the nature of the iterative method.

closed 2017-11-29 by svenboulanger