SpiceSharp/SpiceSharp #16: Question: How to calculate amount of current between pins?

CIVITAS-John, 2017-12-16:
Hi Sven,

Thank you for your hard work first - that's the cornerstone of anything grown out of it. 

And here is my trivial question: how can I calculate amount of current between two connected pins without adding a trivial resistor between them (to simulate the wire between)? I know it works but wonder if Spice or Spice# has already provided some better approaches since every extra components cost performance. 

PS when will the next release become available or can I use any branch beyond master for now?

svenboulanger, 2017-12-16:
Hello John,

Thank you for your question. I'm always happy to see someone interested in our project.

In Spice 3f5, the "official" way of simulating the current through a wire is by adding a voltage source of 0V. Your solution of adding a resistor of will work less well because a resistance of 0 ohm is not allowed (it needs a finite conductance) so it will influence the circuit behavior.
It should be noted that most time is spent on solving the matrix equations. So adding one more resistor or voltage source should not have a big impact.

If you want to use a more up-to-date branch, I suggest using the "development" branch. This branch compiles and passes all unit tests. The main advantage of this branch is that it uses a specialized matrix solver (based on Spice 3f5) that drastically improves performance for larger circuits.
The disadvantage is that it is not yet well documented...

We (me and Marcin) are currently working on refactoring the whole project for version 2.0. It is a big change, so we still need some time to finish it all. It is hard to put a date on it. Some of the features:
- Dedicated sparse matrix library
  - Removes dependency on Math.NET
  - Increases performance drastically
- Behavior-based system
  - Easier to add custom components and models
  - Easier to add custom simulations

CIVITAS-John, 2017-12-17:
Thank you very much for explaining so much and if I am not misunderstanding I shall add a voltage source between those two connected pins with 0V? Sounds somehow not so intuitively lol.

svenboulanger, 2017-12-17:
I admit that it looks more like a hack.
[Spice reference](https://bwrcs.eecs.berkeley.edu/Classes/IcBook/SPICE/UserGuide/elements_fr.html): "zero valued voltage sources may be inserted into the circuit for the purpose of measuring current."

closed 2018-02-04 by svenboulanger