SpiceSharp/SpiceSharp #17: Question: Do I have any way to change some parameters in-simulation?

CIVITAS-John, 2017-12-29:
Dear developer of Spice#,

It's a question which can be solved by 2 ways, while I know neither of them. 
1) Can I pause a transient simulation (or stop it) and resume it with some parameters changed, while others remain the same? (e.g the capacitor has the same voltage as initial condition, the AC source has the same phase, etc.) What if new elements are added or removed?
2) Can I modify some parameters in a transient simulation on-the-fly (not pausing or stopping the simulation, e.g I can inject something when exporting), like the voltage of a capacitor, resistance of a resistor, and so on? 

It's late now and sorry for my chaotic expression yet I believe that I stated the problem clearly enough lol. 

Best Regards

svenboulanger, 2017-12-29:
Hello,

I can confirm that this feature will be part of Spice#. The way it will most likely work, is via an event that is called before trying to solve for a new time point.
I don't plan on adding a dedicated "pause" feature like in Spice 3f5. Simulations can be paused when events are called.
Adding/removing elements during simulation will not be supported. Doing something like that is very costly (all matrix/rhs/solution/old solutions/states need to be reallocated) and may cause convergence issues in some cases (eg. adding a cap with an initial condition causes abrupt voltage changes).

Kind regards,
Sven

CIVITAS-John, 2017-12-30:
Hi Sven,

Can it be done by saving circuit status and re-create a new one? Your proposal is also great and these two ways are hoped be both available. Though it is expensive, but it's necessary when creating a interactive simulator.

Thank you for your reply again. 

Best regards,
John Chen

CIVITAS-John, 2017-12-31:
To be more specific, for example I cannot find a way to simulate capacitor break-down (shall I use something out of RLC?). If I detect the voltage is too high during simulation, no way is present to cut-it-off the circuit..

svenboulanger, 2017-12-31:
You can create a custom capacitor component/model that also models breakdown and/or a flag that enables/disables it. It should be relatively easy to extend Spice# functionality.

Adding or removing components dynamically however are a different story. I will need to think about it some more (what with floating nodes, convergence issues, timestep control, filling in/removing states, etc.). It is not a priority right now.

If Spice# were not to be concerned with accuracy, and if a forward integration method would be used, the problem would actually be trivial. But then we'd just be recreating the educational tool [Falstad](http://www.falstad.com/circuit/) simulator.

CIVITAS-John, 2017-12-31:
Then how can I create a (generic) component that can be disabled or enabled at any time? (I'm currently using Development branch and it has been refactored to behaviors)

Sorry for my ignorance :)

svenboulanger, 2018-01-02:
I believe at this point it would be the easiest to copy-paste the code from the model. Then you add a flag that allows the behavior to be bypassed.

```C#
public class FlaggedBehavior : BehaviorBase
{
    public bool Enabled { get; set; } = true;

    public void BehaviorExecuteMethod(...)
    {
        if (!Enabled)
            return;
        // Original behavior code
    }
}
```

I will suggest not to go too far with this, as the framework will undergo significant changes in the near future.

svenboulanger, 2018-10-08:
As of version 2.5.0 it is possible to modify parameters without having to override behaviors. I wrote some documentation about it here: [https://spicesharp.github.io/SpiceSharp/articles/changing_parameters/example.html](https://spicesharp.github.io/SpiceSharp/articles/changing_parameters/example.html).

The example discusses changing a resistor value during a transient simulation. I believe this should conclude this feature request. Please start a new issue if you find anything more.

closed 2018-10-08 by svenboulanger

chaojian-zhang, 2020-12-26:
> You can create a custom capacitor component/model that also models breakdown and/or a flag that enables/disables it. It should be relatively easy to extend Spice# functionality.
>
> Adding or removing [...]

Can you please elaborate a bit and point out some reference resources on the "forward integration method"? For instance some documents on how that might explain the difference between Falstad and SPICE# implementation?

Thanks!