SpiceSharp/SpiceSharp #18: Oh one more thing: K value of mutual inductance seems to be wrong

CIVITAS-John, 2017-12-31:
Hi Sven,

I've noticed another problem which troubles me a little :) That's the K(MUTcoupling) value of mutual inductance. Right now (using Development branch) it behaves as if MUTcoupling = 0.5 equals to K = 1.0 while MUTcoupling = 1.0 causes unpredictable behavior. I have no ability to investigate it more deeply, and therefore can only submit an issue here.

Thank for your great work again.
Best Regards

CIVITAS-John, 2018-01-01:
Hi.. One more potential bugs:

Connect a circuit with a AC source (sin waveform), a resistor, a capacitor and a inductor in a series and you'll find out that some 'sparse matrix exception' happens. Add a resistor parallel to the capacitor solve this, but the inductor shows zero current which is impossible since the capacitor has. I don't know whether there are any unit tests concerning this and just cannot find one in Development Branch unit tests. 

If needed I can build some codes repeating it; since I am using program to generate circuit it will not give me the C# code at the same time...

CIVITAS-John, 2018-01-01:
Happy New Year :)

svenboulanger, 2018-01-01:
Hello,

The inductor and mutual inductance has not been properly tested. We will add these unit tests soon.
We are working in the refactoring branch right now with some big changes and I can confirm that I found some bugs when using inductors.

Happy new year to you too 😄 
Sven

CIVITAS-John, 2018-01-02:
So can I use Refactoring branch for now? lol

svenboulanger, 2018-01-02:
I believe that will be difficult. The refactoring branch is not ready to be merged with the development branch yet.

CIVITAS-John, 2018-01-02:
Can it pass unit tests? My program is loosely based on Spice# so it won't cost too much.

svenboulanger, 2018-01-02:
It does not pass unit tests because that project also depends (for now) on the SpiceSharp.Parser project.
The ones that I believe I got working right now are: R, L, C, V (independent voltage source) and I (independent current source).

CIVITAS-John, 2018-01-03:
So if I do not use Parser, RLC circuits are behaving correctly in Refactoring branch?

svenboulanger, 2018-02-04:
Unit test has been added for a inductors and mutual inductances.

closed 2018-02-04 by svenboulanger