SpiceSharp/SpiceSharp #20: Voltage source & inductor bug

CIVITAS-John, 2018-01-03:
Hi sven, 

I noticed a strange behaviour where in transient simulation:
1) Connect a voltage source, an inductor and a resistor;
2) Give the voltage source an initial voltage;
3) Start the simulation, and the resistor has a burst in power & current as if the inductor is releasing energy;
4) If we give the voltage source initially zero voltage and something after 1 export, the resistor slowly grows its power as if the inductor is accumulating energy.

I don't know whether it's an Spice-defined behaviour or not so what I can do is to report it here.

svenboulanger, 2018-01-03:
Hi,

It might be a bug. Can you post some code to reproduce the issue?

svenboulanger, 2018-03-27:
It's hard to verify whether or not this is a bug. The project just got refactored as well, so I will be closing this issue.

closed 2018-03-27 by svenboulanger