SpiceSharp/SpiceSharp #21: Unable to use Inductor

neonxc, 2018-01-10:
Hello,
first of all thank you very much for your work. It's a pleasure to work with!

However I encountered a problem I cannot deal with. Whenever I use an inductor and check the circuit a null reference exception is thrown.

I cannot simulate or check even this simple circuit using your 1.1.0 release:

```
Circuit ckt = new Circuit();
      var src = new Voltagesource("V1", "A", "GND", 0);
      src.VSRCacMag.Set(12);
      ckt.Objects.Add(
        src,
        new Resistor("R1", "A", "B", 10),
        new Inductor("L1", "B", "GND", 0.000018));

      ckt.Check();
```

> System.NullReferenceException
>   HResult=0x80004003
>   Source=SpiceSharp
>   StackTrace:
>    at SpiceSharp.Components.Inductor.Setup(Circuit ckt)
>    at SpiceSharp.Circuit.Setup()
>    at SpiceSharp.Circuits.CircuitCheck.Check(Circuit ckt)
>    at SpiceSharp.Circuit.Check()
>    at test.Program.Main(String[] args) in G:\Můj disk\schu\c#\2\test\Program.cs:line 166

What am I doing wrong? Thank you very much.

svenboulanger, 2018-01-10:
Hello,

First of all, thank you for your input!
This is a bug. Inductors and mutual inductances were not properly tested yet for v1 so there are a few issues with them. Right now we are working on Spice# v2 where the issue should be addressed, but the branch is not yet usable. The best I can do is a quick fix in the master branch.

neonxc, 2018-01-10:
Oh thank you very much, Sir! Just one more thing; the subcircuits disappeared since the last release, I relied heavily on them. Is there any easy replacement when creating the circuit programmatically?

svenboulanger, 2018-01-10:
Hello,

The easiest method would probably be by making your own container class for circuit objects I'm afraid. There are no plans for implementing the Subcircuit-class again as it is seen as a part of the parser rather than the simulator.

svenboulanger, 2018-02-04:
Inductors should work now. Closing this ussue.

closed 2018-02-04 by svenboulanger