SpiceSharp/SpiceSharp #52: Current Source - simulations

majdaneq, 2018-10-28:
Hey there, it's me again. 

I'm enjoying using your library. Could you please explain me, what am I doing wrong.

I've got two different circuits:
1)  var ckt = new Circuit(
        new CurrentSource("I1", "in", "0", 0.1),
        new Resistor("R1", "in", "out", 1.0e3),
        new Resistor("R2", "out", "0", 2.0e3)
    );
2)  var ckt2 = new Circuit(
    new CurrentSource("I1", "in", "0", 0.1),
    new Capacitor("C1", "in", "out", 1.0e-6),
    new Resistor("R2", "out", "0", 2.0e3)
);

For the first one I've got a DC Simulation : 
var dc = new DC("dc", "I1", 0.0, 5.0, 0.1);
dc.Run(ckt);
And for the second, the same simulation:
 var dc2 = new DC("dc", "I1", 0.0, 5.0, 0.1);
dc2.Run(ckt2);

The second simulation returns an error : 
SingularException: Singular matrix
SpiceSharp.Algebra.Solve.MarkowitzEntireMatrix`1[T].FindPivot (SpiceSharp.Algebra.Solve.Markowitz`1[T] markowitz, SpiceSharp.Algebra.SparseMatrix`1[T] matrix, System.Int32 eliminationStep) (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)
SpiceSharp.Algebra.Solve.Markowitz`1[T].FindPivot (SpiceSharp.Algebra.SparseMatrix`1[T] matrix, System.Int32 eliminationStep) (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)
SpiceSharp.Algebra.RealSolver.OrderAndFactor () (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)
SpiceSharp.Simulations.BaseSimulation.Iterate (System.Int32 maxIterations) (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)
SpiceSharp.Simulations.DC.Execute () (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)
SpiceSharp.Simulations.Simulation.Run (SpiceSharp.Circuit circuit) (at <4b44f102b9d74f26b64afbe3a4e65f45>:0)

Could you please explain, if I'm doing something wrong or it's a bug. 

One more question from my side - am I possible to include Sin and Pulse in Current Source as in Voltage Source?

Thank you for your answer in advance.
Regards,
Simon

majdaneq, 2018-10-28:
I've checked the similar circuits, RL, LL , LR are working. Unfortunately when Capacitor is placed the simulation returns an error.

majdaneq, 2018-10-28:
Thank you for your reply. 

correct me please if I'm not thinking right way:
When we are using a current source + capacitor in our circuit any of simulations(DC, AC, trans) won't work cause of the floating node (capacitor is just an empty space for the current). 

Could you please tell me if we are able to set Sin and Pulse attribute to the current source?
Will it change the validation of the basic circuit with a capacitor?

Regards, 
Simon

svenboulanger, 2018-10-28:
The simulation will not work for DC or AC analysis, but it is possible to make the circuit work with transient analysis.
You can **force** the capacitor to start at a certain value for transient analysis, like many other components. For this, you will have to specify the `ic` (initial condition) parameter on the capacitor.

So this means the circuit can be constructed using:

```csharp
var ckt2 = new Circuit(
    new CurrentSource("I1", "in", "0", 0.1),
    new Capacitor("C1", "in", "out", 1.0e-6),
    new Resistor("R2", "out", "0", 2.0e3)
);

// Force the capacitor to start at 0V for transient analysis.
ckt2.Entities["C1"].SetParameter("ic", 0.0);
```

Circuit validation will still throw an exception, but the transient analysis will not complain if you add that last line, because it then has help solving the initial state of the circuit.

svenboulanger, 2018-10-28:
The circuit will not work with DC or AC analysis, but it can be made to work with a transient analysis by specifying **initial conditions**.

Specifying a sine or pulse waveform for the current source will not affect validation. Circuit validation will validate the circuit under the assumption that any simulation could be run on it.

majdaneq, 2018-11-02:
Hey,
hope you are going well. I've found the same exception in circuits with 2 Capacitors or 2 Inductors. I'm using a voltage source with Pulse/Sin and want to make trans simulation. Is it also connected with initial conditions? 
Regards,
Simon

svenboulanger, 2018-11-05:
Depending on how you connect them, you may find that you have floating nodes in the circuit, leading to an unsolvable circuit. Similarly, having only inductors in your circuit may short-circuit a voltage source, also leading to an unsolvable circuit.

I believe that capacitors can be solved by using initial conditions on nodes. Inductors are a bit more tricky, as they need to have their current to be started at 0. Look into the "ic" parameter of inductors for this.

commit bbfa8ff8 (2018-10-28): Fix #52