SpiceSharp/SpiceSharp #54: Resistor enhancement

marcin-golebiowski, 2018-11-06:
Please consider adding parameter to constructor of Resistor that specifies that `_temp.Temperature(simulation)` is executed in LoadBehavior of the resistor.
In other words: temperature behavior can be optionally executed on load.

svenboulanger, 2018-11-06:
Could you please elaborate on why that would be more advantageous than calling the temperature behavior in the `BeforeLoad` event or adding your own custom entities?

Each behavior now has a clearly defined function. Temperature behaviors are called when temperature changes, LoadBehavior is called for calculating a DC solution. Inserting a call to `Temperature()` in LoadBehavior would destroy that.

marcin-golebiowski, 2018-11-06:
I will try :)
At the moment resistance is computed in TemperatureBehavior. I want to make a resistor which has a resistance that can change during simulation. It's not very easy to achieve now. That's way I wanted to introduce some kind of hack (e.g. calling Temperature() during load). 

I feel that is not clean and not good solution. But i don't have a better and clear idea.

marcin-golebiowski, 2018-11-06:
@svenboulanger  I will try your ideas.

marcin-golebiowski, 2018-11-07:
@svenboulanger OK. You were right. The changes are not really needed. 
However I would welcome change regarding to getting temperature behavior of specific entity:
I had to write following code:

[image]

svenboulanger, 2018-11-08:
You should be able to get it by using

```csharp
var temp = simulation.EntityBehaviors[entity.Name].Get<BaseTemperatureBehavior>();
```