SpiceSharp/SpiceSharp #60: Adding L,W parameter for MOSFET model

marcin-golebiowski, 2018-11-14:
Is that ok to add l, w parameters to mosfet model? 

Take a look at following part of netlist:

```
M14 12 2 15 15 NMI 
.MODEL NMI NMOS(L=2.00U W=42.0U KP=20.0U LEVEL=1 )

```

svenboulanger, 2018-11-14:
I'm not sure I'm following:

https://github.com/SpiceSharp/SpiceSharp/blob/ab6411ff211ba7d34df5e61de73b8991b6d1c415/SpiceSharp/Components/Semiconductors/MOS/Common/BaseParameters.cs#L33-L49

svenboulanger, 2018-11-14:
Oh I see, you mean on the model.

Are you sure it is taken into account in the reference netlist? It is possible that the Spice simulators will just ignore it.

marcin-golebiowski, 2018-11-15:
@svenboulanger  I will try to verify that today.

marcin-golebiowski, 2018-11-15:
@svenboulanger  I found following reference netlist from MaximIntegrated:
https://www.maximintegrated.com/en/design/tools/modeling-simulation/pspice/MAX16010.FAM
At the end of the netlist there are MOSFET models declarations with different W and L

closed 2019-06-24 by marcin-golebiowski