SpiceSharp/SpiceSharp #78: Question: How to do a single-step simulation? (DC Solver)

szinubuntu, 2019-05-25:
Hi There,

Thanks for you great library, it's really easy to use. One thing troubles me is I couldn't find a way to do a **simple DC solver**. 

According to the [tutorial](https://spicesharp.github.io/SpiceSharp/articles/tutorial/gettingstarted.html#running-a-dc-analysis): "A DC simulation will (**by default**) sweep a voltage or current source value. The result is a DC transfer curve in function of the swept parameter." - is there anyway to **do NOT sweep**, but instead just give the steady state voltage values at the nodes?

For instance, for this circuit:

```
var ckt = new Circuit(
    new VoltageSource("V1", "in", "0", 5.0),
    new Resistor("R1", "in", "out", 1.0e4),
    new Resistor("R2", "out", "0", 2.0e4)
    );
```

What I could do is to initialize the simulation like this `var dc = new DC("DC 1", "V1", 4.0, 5.0, 1.0);` and take the last value as my result, but it just feels so weird. what I want to do is to just get the value at **out** when voltage is 5V.

I couldn't find out how to do that in the API manual, and I am wondering is this available?

Thanks!

svenboulanger, 2019-05-26:
Hi and thank you for taking an interest in Spice#.

I believe that you're looking for the OP simulation. 

```csharp
var sim = new OP("op 1");
```

The simulation searches for the Operating Point of the circuit, which is the DC solution to the circuit without sweeping any parameters.

szinubuntu, 2019-05-28:
Thanks a lot Sven, works like a charm!

Below is the complete code:

```
// Build the circuit
var ckt = new Circuit(
    new VoltageSource("V1", "in", "0", 5.0),
    new Resistor("R1", "in", "out", 1.0e3),
    new Resistor("R2", "out", "0", 2.0e3)
    );
// Create an Operating Point simulation
var dc = new OP("my op");
Export<double> currentExport = new RealPropertyExport(dc, "R1", "i");
dc.ExportSimulationData += (sender, exportDataEventArgs) =>
{
    double voltage = exportDataEventArgs.GetVoltage("out");
    double current = currentExport.Value;
    Console.WriteLine($"Out: {voltage} V, Current: {current} A");
};
// Run the simulation
dc.Run(ckt);

// Per Sven: The simulation searches for the Operating Point of the circuit, 
// which is the DC solution to the circuit without sweeping any parameters.            
```

closed 2019-05-28 by szinubuntu