SpiceSharp/SpiceSharp #80: Question: Issue with modeling

user4githvb, 2019-07-02:
I am trying to model following test circuit and I am confused what would be the way of programming it in SpiceSharp specially values and function that are used here: Exm , Guv.
Please advice

I_M  0 N1 {0.02}
X_TEST N1 0 N2 M_TEST params: factor={4}

.subckt M_TEST m_in m_out v_Out  params: factor=2
 
.func  transf(m) {m*factor+3}

Vmas m_in msx {0}
Exm  msx m_out value={V(v_Out)*V(v_Out)*3}
Hmss mss  0  Vmas 1

Guv 0 v_Out value={transf(V(mss))}
Ruv 0 v_Out 1

.ends

svenboulanger, 2019-07-02:
Thank you for using Spice#.

What you are trying to use is behavioral modeling. In many spice simulators you are able to specify arbitrary functions in text format. This means that you need a parser to execute that arbitrary function expression, which I have deliberately left out of this core package.

As I see it you have three options:

- You can create a custom component that fills the Y-matrix and Rhs-vector, similar to all other components in the package. This is probably the most performant solution, but you will also need an understanding of the technical details of Spice#, Modified Nodal Analysis and Newton-Raphson.
- You can try using [image]. It has some functionality for parsing netlists like the one you showed.
- You can wait for `Spice#.Behavioral`. I am currently working on a library for behavioral modeling.

user4githvb, 2019-07-03:
Thank you for providing Spice# and also for quick answer.
Is it possible to get circuit from parser and merge it in Spice# like subcircuit?
Do you have expected date of the first version of Spice#.Behavioral?

marcin-golebiowski, 2019-07-03:
@user4githvb I think it's possible to get a subciruit from parser and merge it. Please give me two days to try this out first.

svenboulanger, 2019-07-03:
Starting with version 2.7, it is also possible to instantiate a circuit into another circuit like a subcircuit directly from the core package. But unfortunately I haven't updated the documentation yet. You can basically use:

```csharp
var ckt = new Circuit();

// Assume you already have a subckt (also a Circuit object)
ckt.Instantiate(new InstanceData(subckt, "myinstance"));

// If you want to map some subcircuit nodes onto global nodes, you can use:
var instanceData = new ComponentInstanceData(subckt, "mymappedinstance");
instanceData.NodeMap.Add("0", "0");
instanceData.NodeMap.Add("VDD", "VDD");
instanceData.NodeMap.Add("m_in", "N1");
instanceData.NodeMap.Add("m_out", "N2");
ckt.Instantiate(instanceData);
```

`Instantiate()` will clone a circuit and merge the cloned circuit with the target circuit. This is the expected behavior when treating a circuit like a subcircuit. `Merge()` will just merge all entities of a circuit with the target circuit by reference. Unlike instantiating, changing entity parameters in the original circuit will also affect the entities in the merged circuit as they are shared.

But if you intend to use [*Spice#.Parser*](https://github.com/SpiceSharp/SpiceSharpParser), you may just want to use the parser for everything! *Spice#.Behavioral* is actively being developed. I am hoping to publish a first version on GitHub in a week or two.

user4githvb, 2019-07-03:
With SpiceParser I have issue with my example as well as with diode example (from tutorial):

            var netlist = string.Join(Environment.NewLine,
                "Diode circuit",
                "D1 OUT 0 1N914",
                "V1 OUT 0 0",
                ".model 1N914 D(Is=2.52e-9 Rs=0.568 N=1.752 Cjo=4e-12 M=0.4 tt=20e-9)",
                ".DC V1 -1 1 10e-3",
                ".SAVE i(V1)",
                ".END");

            // Parsing part - SpiceSharpParser
            var parser = new SpiceParser();
            var parseResult = parser.ParseNetlist(netlist);
            var spiceSharpModel = parseResult.SpiceSharpModel;

Method not found: 'Boolean SpiceSharp.Circuits.Entity.SetParameter(System.String, Double, System.Collections.Generic.IEqualityComparer`1)'.

marcin-golebiowski, 2019-07-03:
@user4githvb Most likely you are using new version of SpiceSharp. SpiceSharpParser is at the moment compatible with older version. 

Please clean nuget packages, remove SpiceSharp, install SpiceSharpParser again. It should also include SpiceSharp.

user4githvb, 2019-07-03:
That was the issue - but now I will not be able to instanciate subcircuits - I guess I need to wait for  Spice#.Behavioral 
I will just check custom component method but I feel that is quite demanding. Thank you for help

marcin-golebiowski, 2019-07-03:
@user4githvb  I will publish new version of Parser for you. It will include SpiceSparp 2.7.

marcin-golebiowski, 2019-07-03:
@user4githvb I will publish tomorrow. Sorry about that. There are some problems with integration tests.

user4githvb, 2019-07-05:
Now parser works fine - thank You thank You very much.

closed 2019-07-05 by user4githvb

user4githvb, 2019-07-19:
Hello. Any neews on Spice#.Behavioral?

svenboulanger, 2019-07-19:
Hi,

The basic framework for regular voltage and current sources work for DC analysis. It took a little bit more time than expected because there was some extra complexity that I overlooked, but it works since yesterday. I am now working on the other behaviors (time behavior and frequency behavior), which should not take too long. I am hoping to complete it this weekend.

Yours sincerely,
Sven

marcin-golebiowski, 2019-07-19:
@user4githvb Any problems with Spice#.Parser ?

user4githvb, 2019-07-19:
I used this way with parser - I completed proof of concept - I had different result comapred to Spice - I abandoned work due to the fact that we have concluded that maybe better wait for Spice# to evaluate. We wanted not to have Spice Code in the #code due to maintanance reasons.

marcin-golebiowski, 2019-07-19:
@user4githvb Could you so kind and provide more details about your results and the test? 
I would like to fix parser if it's possible to have same results.

svenboulanger, 2019-07-21:
@user4githvb [Spice#.Behavioral](https://github.com/SpiceSharp/SpiceSharpBehavioral) has been created. I should mention that given the complexity of the project, it is possibly unstable. We should test it thoroughly in the future. If you have any issues, don't hesitate to post an issue there.
Once the project stabilizes, I will upload it to NuGet, but fow now please checkout from the repository directly please.