SpiceSharp/SpiceSharp #93: SpiceSharp.CircuitException: Non-convergence error message

rgawry, 2019-09-09:
Hi,
Im learning SpiceSharp + SpiceSharpBehavioral, I created simple parallel circuit without using behavioral elements and it is working fine, when I use some of the elements from SpiceSharpBehavioral like BehavioralCurrentSource I get error "SpiceSharp.CircuitException: 'Non-convergence, node Node m_in (Voltage) is not a number.'", any ideas how to solve this?

svenboulanger, 2019-09-09:
Hi!

The exception you're getting is typical for a circuit that is unsolvable. The simulator will throw this exception if one of the voltages or currents is `double.NaN`.

If you can give an example I might be able to help.

rgawry, 2019-09-10:
hi, i found out that i have problem with behaviour expression, i need to use pwr(x,y) function in expression and spice does not recognize it when parsing expression, i can see that in spiceSharpBehaviour code there are defined math functions like pwr with name pow, but neither pow or pwr works, so i tried to implement that pow funciotn using FunctionFound mechanism, i could get rid of expression parsing error but I end with non convergence/voiltage in NAN error.
in conclusion i need to use pwr function in expression

svenboulanger, 2019-09-10:
You can specify a power in SpiceSharp.Behavioral using `Pow()`, it is case sensitive.

Specifying your own custom functions requires more work. I suggest looking at or using the helper classes in the `SpiceSharpBehavioral.Parsers.Helpers` namespace to see how you are supposed to implement your own custom methods.

An example that works right of the box:

```csharp
using System;
using SpiceSharp;
using SpiceSharp.Components;
using SpiceSharp.Simulations;

namespace Benchmark
{
    class Program
    {
        static void Main(string[] args)
        {
            var ckt = new Circuit(
                new VoltageSource("V1", "in", "0", 1),
                new Resistor("R1", "in", "0", 1),
                new BehavioralVoltageSource("E1", "out", "0", "Pow(V(in),3)"));

            var op = new DC("dc", "V1", -1, 1, 0.1);
            var export = new RealCurrentExport(op, "V1");
            op.ExportSimulationData += (sender, e) =>
            {
                Console.Write(export.Value);
                Console.Write(" -> ");
                Console.WriteLine(e.GetVoltage("out"));
            };
            op.Run(ckt);
            Console.ReadKey();
        }
    }
}
```

I will close this issue because it doesn't relate to Spice#, but Spice#.Behavioral. Please open an issue [here](https://github.com/SpiceSharp/SpiceSharpBehavioral/issues) instead if you have any questions left about Spice#.Behavioral.

closed 2019-09-10 by svenboulanger