SpiceSharp/SpiceSharp #96: Circuit Validation Results

ToddThomson, 2019-12-20:
Similarly to the ```Parser.ParseNetlist()``` being able to return error results rather than exceptions, it would be nice to be able to obtain the full set of Circuit Validation issues. This would allow the end-user to have greater detail on what needs to be corrected before running simulations. See issue SpiceSharp/SpiceSharpParser#142

ToddThomson, 2019-12-20:
@marcin-golebiowski This and issue #97 are quite important to me for work on the SpiceSharp Netlist Simulator for UWP that I want to complete over the holidays.

svenboulanger, 2019-12-21:
v3.0 will have a system with a "validator" that has multiple "rules". If a rule is violated, the validator calls an event. It then allows you to ignore the violation which keeps them from raising an exception (so if no methods are handling those events, an exception is thrown by default).

The circuit `Validate()` method can be removed. I was currently thinking more along the lines of a "validator" validating a circuit, rather than a circuit that can be validated.

Would this be agreeable or would you like to see something different?

svenboulanger, 2019-12-31:
v3.0 will allow you to run validation on circuits and enumerate all violations without raising an exception.

Simulations will also have their own validation that they will run before running the simulation. If that validation fails, then an exception is raised (as it would definitely raise an exception later on anyway). This standard validation can be switched off if performance is an issue (or if your own validation is already adequate).

svenboulanger, 2020-11-28:
Validation is supported since v3.0.

closed 2020-11-28 by svenboulanger