SpiceSharp/SpiceSharp #97: Running a Simulation with an Invalid Circuit

ToddThomson, 2019-12-20:
I haven't fully worked my way through Circuit construction, Circuit validation and Circuit simulation, but I am not clear on why a Circuit does not have an IsValid property. The Circuit class does have a Validate() method, but all it does is return a CircuitException if a validation rule fails ( #96 ).
Perhaps the Circuit class could contain an IsValid property that calls a private Validate() method when the circuit entity items have changed ( a needs validation flag ). This could then be used by the ```Simulation.Run( entities )``` method to check if a circuit is valid before running simulations.

svenboulanger, 2019-12-21:
The original validation was quite naive and straight-forward. I also chose the form of a method (`Validate()`) to make you feel that it was not a very cheap operation. But I do believe that making it throw exceptions for any rule violation was perhaps not the best option because it obscures validation rules further down the line.

The new validation framework (for v3.0) should allow you to run through all the validation rules and list them.

svenboulanger, 2019-12-23:
So, I've been thinking about this...

The question stuck in my head is "what is a valid circuit"? And the answer is actually a little bit more awkward than I assumed before. For the standard simulations included in Spice#, some rules apply and some don't. For example, floating nodes are invalid for biasing analyses, but may not be for a transient analysis that has initial conditions. For custom simulations, I could conceive types of simulations that don't even require a ground node or independent source.

So a valid circuit would depend on the context and an `IsValid` flag would need a lot of assumptions (which is currently the case by the way). I'm considering to remove the `Validate()` method altogether.

svenboulanger, 2020-11-28:
This has been dealt with using circuit validators.

closed 2020-11-28 by svenboulanger