SpiceSharp/SpiceSharp #98: Question: Maximum reasonable circuit size

RicParPar, 2020-04-03:
Hi,

I am currently working on a project using Spice#. My goal is to make a DC simulation of some big circuits (hundreds of resistors + hundreds of diodes + tenths of transistors + a couple of sources to drive a transistor). I would like to obtain a couple of operating points. After creating and validating a circuit with the previous characteristics, the simulations returns the "Could not determine operating point" exception. After increasing the number of iterations the result was the same and, when I lowered the Gmin parameter to 1e-18, the simulation returned the "Singular matrix" exception. Now I am wondering if there are any other parameters I can tweak to make it converge (like tolerances), or if the circuit is just too big to be handled by the library. In the latter case, do you have an idea of the maximum number of components it is reasonable to simulate?

Thank you in advance.

Regards,

Ricardo

svenboulanger, 2020-04-03:
Hello!

## Condition number/pivoting
By my knowledge, the size of the circuit should not matter for the errors that you're getting, but the **condition number of the matrix** can, as well as the pivoting scheme. This basically means that a small change at some critical points in the circuit (like a numerical error) will cause a large change to the solution, hindering convergence.

This problem usually occurs at extremely high-impedance nodes, for example at the output of a mosfet of which the output impedance is extremely high. In that case, reducing the mosfet output impedance will greatly improve the sensitivity. Pivoting the matrix is done relatively conservatively to improve speed, but you can control the pivoting scheme by making the pivoting tolerances stricter. Ie. reduce` RelativePivotThreshold` and `AbsolutePivotThreshold`, as this will make the solver optimize the accuracy more by pivoting the matrix more regularily.

## Badly behaving models/gmin
Another issue that can creep up is just **the model**. "Problematic" devices that are known to cause extremely slow convergence are typically diodes and transistors (due to their exponential curves). If these exponential devices are the ones causing issues, then the first parameter to look at would be the `Gmin` property. The `Gmin` parameter is a conductance added in parallel to each exponential curve to help convergence. You should increase `Gmin` to help convergence.

## Iteration tolerances
Relaxing tolerance can help, but you just have to take into consideration that the results will be less accurate too. The simulator will just stop simulating once the solution doesn't vary more than the allowed tolerance.

## Helping convergence before simulating
What I also think could help greatly in your project, is that you can help the simulator by specifying nodesets. These are solutions that will be "close to ...", especially at nodes that you think are tricky to find (eg. if there is some kind of very harsh nonlinear curve involved, like a close-to-ideal switch).

## Breakdown
Given that you have a result (although it is an exception), is to me a strong indication that the number of components/nodes does not play a significant role. What seems much more likely to me is that there is one or more points in your circuits that are badly defined. Linear circuits (RLCM components and any sources) are usually not an issue (unless you're using crazy values), so start looking for and around the very nonlinear devices.

You can also try accessing the [`ProblemVariable` property]https://github.com/SpiceSharp/SpiceSharp/blob/e4fee848a3940344f0f753d079f784370b43e65c/SpiceSharp/Simulations/Base/BaseSimulation.cs#L29) on the simulation. This property is always set to the variable that did not meet the specified tolerances. You can register for the `BeforeLoad` event if you want to figure that out for each iteration, or just read it after simulation if you want a quick starting point.

RicParPar, 2020-04-06:
First of all, thank you very much for your quick response. In order to make a quick test, I just defined all my transistors and diodes with the same simple models. All those models were left with their default parameters except for the following ones:
```
		// Add models
		var diodeModel = new DiodeModel("GenericDiode");
		diodeModel.SetParameter("vj", 0.7);
		var tPnpModel = new BipolarJunctionTransistorModel ("GenericPNP");
		tPnpModel.SetParameter("pnp", true);
		tPnpModel.SetParameter("bf", 350.0);
		var tNpnModel = new BipolarJunctionTransistorModel("GenericNPN");
		tNpnModel.SetParameter("npn", true);
		tNpnModel.SetParameter("bf", 350.0);
		var pFetModel = new Mosfet1Model("GenericP-FET");
		pFetModel.SetParameter("pmos", true);
		pFetModel.SetParameter("kp", 150.0e-3);
		var nFetModel = new Mosfet1Model("GenericN-FET");
		nFetModel.SetParameter("nmos", true);
		nFetModel.SetParameter("kp", 150.0e-3);
		spiceCircuit.Add(diodeModel, tPnpModel, tNpnModel, pFetModel, nFetModel);
```
Then, some sources were defined to drive one of the transistors in the circuit. For example, for NPN Bipolar transistors, the sources are configured like this:
```
		var csBaseNpnT = new CurrentSource("Is_B", "GND", baseN.GetSpiceName(), 1.0e-3);
		var csBaseNpnTRes = new Resistor("RES_Is_B", "GND", baseN.GetSpiceName(), 1.0e9);
		var vsEmitterNpnT = new VoltageSource("Vs_E", emitterN.GetSpiceName(), "GND", 0);
		var vsCollectorNpnT = new VoltageSource("Vs_C", "[Vs_C_POS]", "GND", 5);
		var vsCollectorNpnTRes = new Resistor("RES_Vs_C", collectorN.GetSpiceName(), "[Vs_C_POS]", 1.0e-3);
```
That is, a current source is connected to the base of the transistor, together with a big resistor in parallel to avoid singularities. A voltage source is connected to the collector of the transistor with a small resistor in series to avoid singularities. To conclude, a 0 V voltage source is connected to the emitter to short this net to the simulator GND. The simulation is then executed as follows:
```
		// Define DC simulation for NPN transistors
		var dcNpnT = new DC("DC_TNPN", new[]
		{
			new SweepConfiguration("Is_B", 0.0, 100.0e-6, 100.0e-6),
		});
```
Using the previous code, I could obtain satisfactory results in small circuits with 10 to 20 components. That's the reason why, at first, I thought that the exception could be just a result of the increase in the number of components. Do you think that a better definition of the models would improve the convergence? So far I have tweaked the value of `Gmin`, `DcMaxIterations` and other simulation parameters with no luck. I have also followed your instructions to see the `ProblemVariable` and I could detect that the simulation has problems with different nets in different iterations. In most of the cases, the problematic net is [GND] (Voltage). Lastly, I am not completely sure about how to modify the pivot thresholds and specify `NodeSets`.

svenboulanger, 2020-04-06:
Hi,

> I have also followed your instructions to see the ProblemVariable and I could detect that the simulation has problems with different nets in different iterations. In most of the cases, the problematic net is [GND] (Voltage).

Is this supposed to be a "real" ground node (ie. short-circuited to the simulation ground node - "0" or "GND"). If this node is supposed to be the ground node, then you may have accidentally created a node called "[GND]" that isn't actually connected to the simulation ground. It would explain a lot of your errors, so please double check that this isn't an error.

> I just defined all my transistors and diodes with the same simple models

It is actually the most simple mosfet models that can potentially cause the most issues. The models that you specified (`Mosfet1` and `Mosfet1Model`) have an infinite output impedance by default and the `Gmin` parameter doesn't have any effect on that (it means that they can potentially work like ideal current sources causing floating nodes, and the simulator can't detect this until it is too late). I suggest setting the `Lambda` parameter on the `Mosfet1Model` to something higher (eg. 1e-5).

The BJT can cause numerical instability due to its exponential curves, but increasing `Gmin` should help here.

> Do you think that a better definition of the models would improve the convergence?

That is hard to say. I would need a closer look for that.

> Lastly, I am not completely sure about how to modify the pivot thresholds and specify NodeSets.

Every iteration, each component/model will fill up a matrix with equation values that the simulation will then solve in one go. In order to solve that matrix, it needs to reorder the matrix to minimize numerical errors, but this is a costly operation so the framework tries to limit reordering by allowing some numeric error through the `AbsolutePivotThreshold` and `RelativePivotThreshold`. If you make these values very small, then the matrix will be reordered more often and numerically the result should be better, but the simulation might take a bit longer.

Nodesets are just initial guesses that you know upfront. You can set them by adding values to the dictionary of the `BaseConfiguration` of the simulation. If you know that the voltage at node "A" is going to be around 2V, then you can tell the simulator by writing:

```csharp
var op = new OP("op");
op.Configurations.Get<BaseConfiguration>().Nodesets.Add("A", 2.0);
```

The simulation will try to have a **first** iteration with the voltage of A as close as possible to 2V while filling in the rest. Then it lets the simulator loose and finds the real solution. The advantage here is that you can make the simulation converge faster to a solution because it can start from a better starting point.

If the above fails, would you be okay with sharing your project with me such that I can have a closer look?

Yours sincerely,
Sven

RicParPar, 2020-04-07:
Hi,

Thank you very much again for your clear answer. To give you some context, I am doing this project for my master thesis and it will be part of a bigger software that is being developed at a company. For that reason, it is complicated for me to share code straight away and I always have to ask my supervisors first. Today they told me that it would not be a problem to extract the piece of code that performs the simulation and show it to you together with the netlist of the circuit that I am using for debugging. I will try to do that this week.
With respect to the GND question, [GND] is a node that corresponds to the reference of the original circuit that I am parsing. Then I have the real GND node of the SPICE circuit that I am simulating. So that should not be the issue.
Today I also made the results converge by including the 1e-5 lambda value on FET transistors and increasing Gmin to 3.1e-5. However, the results are not the expected ones yet, since some measured currents were much higher than expected (I still have to see if this could be due to a bug in the SPICE parsing routine). Other values of Gmin lower than that make the solution to not converge.

Best regards,

Ricardo

svenboulanger, 2020-04-07:
Hi Ricardo,

If possible I'd like to take a look to the circuit. You can contact me via svenboulanger@hotmail.com.

Yours sincerely,
Sven

svenboulanger, 2020-04-16:
After looking at the circuit, I discovered that the problem was most likely a **resistor of 0<img src="https://render.githubusercontent.com/render/math?math=\Omega">**. Simulators tend to limit the minimum resistance, because they use the conductance <img src="https://render.githubusercontent.com/render/math?math=G = \frac{1}{R}"> to contribute to the equation matrix (Y-matrix).

In Spice#'s case, the minimum resistance is <img src="https://render.githubusercontent.com/render/math?math=10^{-12}\Omega">, causing very large values to appear in the Y-matrix. This is is a problem for numerical stability (the condition number becomes much worse), in turn causing problems for convergence.

closed 2020-04-16 by svenboulanger