SpiceSharp/SpiceSharp #102: Exporting netlist

csculley, 2020-06-16:
Hello, how feasible is it to generate a netlist from a SpiceSharp Circuit? Is this already implemented? It would seem to be just a matter of iterating through all the Entities of the circuit and printing each. Thanks!

svenboulanger, 2020-06-16:
Hi @csculley ,

I don't have plans or the intention to implement exporting to a netlists in the core package. However, I think it is quite feasible to do so (perhaps as an extra package) because in its most simple form it indeed just involves listing all entities and their parameters using reflection (all parameters are tagged with their Spice-equivalent names anyway).

Some small adaptations are probably needed, like extracting a *PULSE* statement from a *Pulse* instance, or what you want to with `VoltageDelay` which doesn't usually exist in most Spice simulators. With v3, there is also the new `Subcircuit` class, and the new `ParallelComponent` class (for allowing parallel computations), and they also don't necessarily translate in a simple way to "standard" Spice netlists.

@marcin-golebiowski is already dealing with a lot by trying to support the syntax of as many Spice-simulators as possible in Spice#.Parser. For example, behavioral sources have different syntax styles in different simulators, so maybe he can also give information if you decide on having a go. Let us know if you have questions getting this to work.

tldr;
I have no intention on working on exporting netlists in the core package. You can always make your own package to suit your own netlist syntax.

csculley, 2020-06-16:
Thanks for the response! While using the SpiceSharp library, I've made my own debug statements just to get an idea of what I'm throwing into the circuit, but found myself wondering/wishing for something more formal to debug further, and wasn't sure if it was implemented already.

svenboulanger, 2020-06-17:
Hi,

An improved debugging experience **is** within my current scope, so I'll reopen this issue and see what we can do from that perspective.

What I'd like to avoid is implementing a netlist exporter that is compatible with other Spice packages. Ie. I want to avoid having logic in the core package that is required to have a netlist in "LTSpice" format, or "HSpice" format.

How about this?:
- Change the ToString() method of entities to include some more information (like the name, and the connections for `Component`).
- Add functionality to list parameters and entity collections through extension methods.
- The resulting extension methods may generate text similar to netlists, but I will not make promises about compatibility with existing Spice packages.

csculley, 2020-08-12:
Hey, I just wanted to say a huge thank you for the advanced logging features! We've already implemented it from the v3.0.1 release and it's been super helpful for debugging. The project I'm working on is actually able to be open-sourced, so I'll try to make sure it gets linked as a dependent when it gets open-sourced soon. Thanks again!

closed 2020-08-12 by csculley

svenboulanger, 2020-08-13:
That makes me happy and motivated to work on the project :-)

csculley, 2020-08-24:
Unfortunately, since the project is hosted on an enterprise instance of Github, it isn't detected as a dependency :pensive:. If you'd like to check out the project, there's a public release available [here](https://github.rcac.purdue.edu/EnvisionCenter/eleclabs-public), but it's still under active development. Releases can be found [here](https://envision-public-builds.s3.amazonaws.com/ElecLab/index.html).

svenboulanger, 2020-08-24:
That's looking pretty nice! I don't mind the dependency not being detected, it's just nice knowing that it's being used. If you find any bugs, please let me know.