SpiceSharp/SpiceSharp #105: ECS

SirMetathyst, 2020-07-26:
Hi, I'd like to use this software in an ECS environment. Would this require modification of any core code or can I simply use what is available as lego pieces to adapt this to fit ECS architecture?

svenboulanger, 2020-07-26:
Hi,

From what I can understand from ECS, Spice# does not strictly follow these principles. Could you explain how you want to split Spice# into lego pieces?

To avoid confusion: I am using Entities and Components as something completely different than ECS (an entity can generate behaviors for a simulation, while a component is a special kind of entity that can be connected to other components - i.e. an IComponent inherits from IEntity).

SirMetathyst, 2020-07-26:
I'm not sure how I want to split it yet but I know that the component behaviour parts will need to be part of a system and I will need components with any necessary data for those behaviours to work on an entity. I intend to use UnityECS so I know that you can embed an "Entity" into a component to link an entity to another to grab extra data through that reference while in a system processing a particular kind of entity makeup in batch.

I think I'll just have to try adapting it and find out how it would even work....

closed 2020-07-27 by SirMetathyst