SpiceSharp/SpiceSharp #113: Tickable Transient Simulation

dsonbill, 2020-10-16:
I'd like to simulate a circuit at specified millisecond intervals to use this in a project. I've done this manually with an old version of SpiceSharp, but things are changing too much for my old TransientTickable class to work any more. Is there any way we could get a Transient.Tick(milliseconds) function?

svenboulanger, 2020-10-16:
Hi,

There are a few options, and each one has its pro's and con's:

- The `FixedEuler` integration method can be passed to a transient simulation ([example](https://github.com/SpiceSharp/SpiceSharp/blob/235f1310d99b8e1dba1231b1bcae5739dffb84e9/SpiceSharpTest/Simulations/TransientTests.cs#L115)). This integration method is by far the fastest type of transient simulation, but it has the worst accuracy (better if the timestep is small), and it exhibits artificial damping (the algorithm causes circuits to be more "stable" than in real life, again better if the timestep is small). Nevertheless, its timestep is fixed and it should work out of the box (e.g. you can specify timesteps of 1e-4 and you only use every 10th point).
- Another way to get points at fixed times is by **interpolation**. This technique is for example used by transmission lines or voltage-delays. The disadvantage is that it might give some unwanted artifacts in some cases, like if you want to determine the frequency-spectrum of the signal. The advantage is that the simulator will keep its accuracy and will not artificially dampen (in case you're using the Trapezoidal integration method).
- The most accurate results will probably involve controlling the **truncation of timesteps** to gently force the simulator to hit certain timepoints. This component can then expose whether or not a timepoint is the one you're interested in. This can be done by implementing a component that you add to your circuit that implements `ITruncatingBehavior` for example.

If you need help with any of these options, please let me know.

svenboulanger, 2020-12-25:
A `Sampler` entity (the third option) is planned to be added for v3.1.0.

svenboulanger, 2021-01-24:
Due to some bug fixes that had to be done, I added the `Sampler` entity to v3.0.4 a little earlier than expected.

I believe this should solve the issue, but if you have any issues we can always reopen it! :-)

closed 2021-01-24 by svenboulanger