SpiceSharp/SpiceSharp #115: ArgumentException When Adding Diode to Circuit

dsonbill, 2020-11-07:
Using the latest 3.0 release, I get the following exception when trying to add a diode to a circuit:

`ArgumentException: The parameter or property 'cd' points to both Current and Capacitance.`

Note that I'm not even setting a parameter called 'cd'.

svenboulanger, 2020-11-07:
Hello @dsonbill ,

Thank you for reporting this issue. They aren't 'parameters' but they are 'properties' - i.e. simulation output. I can see that there are two properties with the same name:
- The current through the diode in pretty much any analysis
- The junction capacitance of the diode in AC small signal analysis

If you were to run an AC analysis using a diode and then tried to ask for the property `cd` of type `double`, it cannot decide between these two properties and Spice# throws that exception.

I referenced the original Spice 3f5 library, and confirmed that they called the 'diode current' `c` instead of `cd`. I will bring out a new version shortly.

svenboulanger, 2020-11-07:
The issue should be resolved with v3.0.2.
This bug was verified and then resolved by [this test](https://github.com/SpiceSharp/SpiceSharp/blob/f4cf32e49ade09a1ea606a44d77396de47da8b62/SpiceSharpTest/Models/Semiconductors/DIO/DiodeTests.cs#L99).

closed 2020-11-07 by svenboulanger