SpiceSharp/SpiceSharp #119: Capacitor in Series leads to a Rule Violation

jaminSch, 2020-11-10:
Hello, first of all thank you for all the effort you are putting into this!

I have an issue with Capacitors:
In a simple Circuit (VoltageSource with Pulse, a Resistor and two Capacitor in Series) i get an "Floating Node Rule Violation". This only happens if i have two Capacitors. If i change the second Capacitor and replace it with a Resistor the error wont occur. Am i doing something wrong or is it a Bug?

Thank you!

The Circuit leading to the error:
[image]

The RuleViolation:
[image]

closed 2020-11-10 by jaminSch

svenboulanger, 2020-11-10:
Hi,

I assume that you found the error, but I'll leave this here for people that might run into a similar issue:

Building a capacitive voltage divider leaves the node between the two capacitors "floating" (node `3` in your example). Any voltage that the simulator assigns to that node will result in a valid solution so it doesn't know what to do with that voltage...

In most cases, this is only a problem for DC analysis and any other analysis that needs a DC solution. It should be possible to assign a voltage manually using initial conditions and bypass the issue and get the simulation to run.

Note that other Spice simulators "automatically" add a resistance to ground of 10^12 ohm to any floating node it encounters. If you want to get similar results to other Spice simulators, simply add a resistor from `3` to `0` with the resistance set to 1e12.

jaminSch, 2020-11-11:
Hi,
Thank you for your quick response! I tried your solution and it works perfectly.