SpiceSharp/SpiceSharp #124: Preconditions are not considered for calculating the Operation Point

jaminSch, 2020-11-19:
Hello, I've got a question about how Preconditions are considered for the Operation Point.

If I give my capacitor in a simple circuit (Voltage Source with Pulse, Resistor, Capacitor) a Precondition of 4.0 Volt and set the simulation.TimeParameter.UseIc flag on true in my understanding the precondition should be considered or am I missing something? What happens in the Code is following:

Simulation.Run() -> Execute() ->

[image]
Before calculating the operation Point it is set that the Preconditions should be considered (Debugging showed that the value _time.UseIc = true in my case).

[image]
While Loading the Behaviors the capacitor is ignored because of the DC Analysis and the UseIc-flag is not considered.

[image]
After finding the Operationpoint without considering the preconditions the flag saying Preconditions should be considered is reset and the InitializeStates() Method is called.

[image]
During this initialization preconditions would be considered if the flag wasnt reset before.

Thank you for your help and for sharing the code it really helps me understanding the computer based analysis of electronic circuits!

svenboulanger, 2020-11-19:
Hi,

Spice has a pretty (in my opinion) confusing way of handling initial conditions. I believe that what you did was specify an initial condition on the capacitor, while in fact you probably need to specify an initial condition on the node.

Check [this example](https://github.com/SpiceSharp/SpiceSharp/blob/4b684a7f60e1aebaa0c41270620e151fa7614906/SpiceSharpTest/Simulations/TransientTests.cs#L132) to see how a "floating node" can be fixed using initial conditions.

Another - less used - alternative way of using initial conditions is using the `UseIc` flag which should skip any DC analysis and will instead use the "ic"-parameters specified on all devices. For example, diodes and BJT's have "ic"-parameters, just like a capacitor.
I don't think the latter feature is tested as it is (from experience) an uncommon thing to use. I may add some tests to be sure in the future.

jaminSch, 2020-11-20:
Hi,
thank you for your help!
I tried your example and it works.

I am still wondering why there is an option to initialize the capacitor when it doesnt change its behavior.
In my last picture you can see the ITimeBehavior.InitializeStates-Method in which the Initial Conditions of the Capacitor are considered if the UseIc flag is set (which is hard reset right before the method call (Picture 3)).

I actually think the UseIc flag doesnt stop the DC analysis as you can see in my code snippets (Picture 1) above, the UseDC flag is set to true  independently of the UseIc flag. 

The topic can still be considered "closed" i guess because your solution with initializing the node does work!

svenboulanger, 2020-11-20:
Thanks for checking!

I think this would be a good time to take a look at the `UseIc` behavior and write some tests and possibly fix the bugs if it doesn't work as expected. I'll leave this issue open for now.

svenboulanger, 2021-01-24:
I've given the UIC behavior an overhaul, but it still needs tests to make sure everything operates as expected. I would like to keep this issue open for a little longer to make sure I add some tests.

closed 2023-12-23 by svenboulanger