SpiceSharp/SpiceSharp #127: Determine Stepsize of Transient Analysis

jaminSch, 2020-11-27:
Hi,
is there a possibility to dertermine the stepsize of the transient analysis?
I thought the "step" variable in the constructor would do that but i tried Stepsize "double.MaxValue", "1e-127" and even "double.MinValue" and it didnt change the resolution of the results at all.
In your Tutorial "Getting started/Transient analysis/" you say that the given step size is used only to calculate the initial time step of the simulation. Does this mean i don't have the opportunity to set the real timestep?

Kind regards and once again thank you for your work and the really fast replys you are giving!

svenboulanger, 2020-11-27:
Hi!

Spice# (like all Spice simulators) uses a variable timestep which is calculated depending on the "error" it thinks it is making. To enforce a strict maximum timestep you can set the `MaxStep` property instead of just the initial `Step` property.

jaminSch, 2020-11-27:
Thank you for your help!

closed 2020-11-27 by jaminSch