SpiceSharp/SpiceSharp #132: [Question][Feature Request] Is there a way to continuously simulate?

chaojian-zhang, 2020-12-26:
Hi, I was just wondering about something similar to #4 and #17 - in my case, I just wished to **pause the transient simulation then perform some logic check before deciding whether the simulation should continue**. I have checked https://github.com/SpiceSharp/SpiceSharp/issues/17 but the existing API seems too complicated for this task (and makes code looks less clean if I shall approach it this way).

To elaborate a bit: In case of transient analysis, I wish to implement a custom logic at each "*time interval*" (_might be different from simulation `timestep` parameter_), then decide whether or not the simulation should continue, that is, instead of specifying a duration (`endtime`) upfront, I would like the simulation to keep running as long as an ending condition is not met.

There are two schemes of this: 1) The simulation just keeps running until some ending condition is set, e.g. through a call back returning false or some conditional switch variable, 2) The simulation runs for a specified interval then checks whether it should continue running.

Either way, I would like to be able to **perform some logic and actions (irrelevant to the simulation itself, for instance, I might want to log all the intermediate stages) in between simulation intervals**. Currently `ExportSimulationData()` works only for retrieving values but it cannot extend or stop the simulation.

I saw a function `Rerun()` on `Transient` class but judging from the Tests (e.g. [ACTests.cs](SpiceSharpTest/Simulations/ACTests.cs) it seems it just reruns the original simulation thus not suitable for this purpose. Also the documentation for `Rerun()` is a bit ambiguous: "Reruns the `ISimulation` with the previous behaviors." - I am not sure what this "previous behaviors" mean - I suppose this means `IBehavior` but is this some sort of implementation-level detail or it's relevant as simulation terminology?

svenboulanger, 2020-12-26:
Hi! Always nice to have some feedback about Spice#! 

> I just wished to pause the transient simulation then perform some logic check before deciding whether the simulation should continue.

I think it should be possible to achieve something like that as follows:

```csharp
// Create the transient analysis, and have it run indefinitely (for now)
var tran = new Transient("tran", 1e-6, double.PositiveInfinity);

// Logic for shortening the transient analysis
tran.ExportSimulationData += (sender, args) =>
{
    if (logic == true)
        tran.TimeParameters.StopTime = 0.0; // The transient analysis will finish if the current time > StopTime
}
```

Note that you can take all the time you want inside the `ExportSimulationData`-event. The transient analysis will just pause until all registered events have finished doing their thing, and only then it will continue computing the next time point.

> I wish to implement a custom logic at each "time interval" (might be different from simulation timestep parameter)

The example I gave does not guarantee you to "hit" certain timepoints (issue #113). Timesteps are - as you may know - handled by controlling the so-called truncation error. If you are interested specifically in the results at specific time points, you can manually add breakpoints (`Trapezoidal` and `Gear` implement a breakpoint system), or you can wait for v3.1 and make use of the `Sampler`-entity to avoid the complexity associated with it.

The breakpoint system would look something like this:

```csharp
// Logic for shortening the transient analysis
tran.ExportSimulationData += (sender, args) =>
{
    var breakpoints = tran.GetState<IBreakpointMethod>().Breakpoints;

    // Add breakpoints where needed if you want to hit certain time points
    // Do logic
    if (logic == true)
        tran.TimeParameters.StopTime = 0.0;
}
```

> I saw a function `Rerun()` on Transient class

The `Rerun()`-method will skip the behavior creation process and immediately restart the simulation. The particular use-case was where you would want to change a few parameters of some components (and in the future even add/remove behaviors), and just run it all again.

For a more elaborate discussion of what entities, behaviors and simulations are supposed to do, I would like to refer to [here](https://spicesharp.github.io/SpiceSharp/articles/structure/flow.html).

svenboulanger, 2021-01-24:
Version 3.0.4 should have introduced the `Sampler` entity, which allows you to "stop" at certain time intervals.

Nevertheless, considering the thumbs up on my previous post, I believe I can close the issue. Perhaps we can create a discussion around this to keep some kind of reference, as this seems to be a recurring issue.

closed 2021-01-24 by svenboulanger