SpiceSharp/SpiceSharp #134: NullReferenceException when calling dc.Run() in 3.0.x versions

BlackBerry007, 2021-01-24:
Hi, 
I used Spice# version 2.X in a project with no problem, now I decided to update to the never 3.0.X versions and no matter what I do, I always get an NullReferenceException (see attached image). Even with an empty circuit (attached code sample) I got the same error. As it says, the Error happens in SpiceSharp.Algebra.Element CompileDefaultAddition() function, but I dont understand what is going wrong here. 
Does anybody can help me with this?
Thank you very much in advance

[image]
[image]

svenboulanger, 2021-01-24:
Hi!

In order to support different types for solving equations, Spice# uses generic implementations for solvers. It creates a generic method for addition/subtraction of two values during runtime. It does this by creating a dynamic method (it creates instruction level code to optimize performance).

In your case, this process fails. This is likely due to the target platform that you're using, which I believe may require ahead-of-time compilation (AoT comilation).

Unfortunately, I *thought* I had made this overridable to avoid the issue, but it turns out that I didn't... I will be publishing v3.0.4 shortly which I believe might fix the issue.

svenboulanger, 2021-01-24:
Version 3.0.4 has been published. Please give the update a roll once it is validated and please give a shout here if it works :-)

BlackBerry007, 2021-01-25:
Thank you for the fast reply and the new version :-) 
Indeed you guessed right, I am using this wonderfull library on the Microsoft Hololens 2 (Windows 10 on ARM64 architecture) for analysis and visualisation of electrical circuits. The error above has been fixed, but another Error occurs in a simple circuit like this:

```
       // Build the circuit
        var ckt = new Circuit(
            new VoltageSource("V1", "in", "0", 9.0), // 9 V
            new Resistor("R1", "in", "0", 100.0)    // 100 Ohm
        );

        // Create a OP simulation
        var dc = new OP("DC 1");

        // Create exports
        var currentExport_v = new RealPropertyExport(dc, "V1", "i");
        var currentExport_r = new RealPropertyExport(dc, "R1", "i");

        var voltageExport_v = new RealVoltageExport(dc, "in", "0");

        double c1 = 0, c2 = 0, v1 = 0;

        // Catch exported data
        dc.ExportSimulationData += (sender, args) =>
        {
            c1 = currentExport_v.Value;
            c2 = currentExport_r.Value;

            v1 = voltageExport_v.Value;
        };

        dc.Run(ckt);
```
The error is attached. I believe this is also problem of the hololens platform, because I get no errors running this on PC...

[image]

svenboulanger, 2021-01-25:
Thanks for trying!

You are the first feedback I have on trying to use this library with AoT compilation, so I'd like to grab this chance and try to make it work.

Spice# uses reflection to list and link all entities and behaviors together. It also creates IL code in a lot of places to avoid the performance drop of having to use reflection every time.

I will need some more time to find and go through all the parts that create IL code, and see if I can bypass it if necessary.

BlackBerry007, 2021-01-25:
thank you for doing this! Until you upload the new version I will use the older 2.x one...
Just another quick question: The older versions work on the hololens, with one drawback. I can retrieve voltages with no problem, but with the sample code above I cant retrieve any current information. It always outputs 0A for every entity. The same code on PC however does output the right currents.
Could this be also a Hololens platform problem?

Thank you very much for your quick and helpful responses!

svenboulanger, 2021-01-27:
> Just another quick question: The older versions work on the hololens, with one drawback. I can retrieve voltages with no problem, but with the sample code above I cant retrieve any current information.

Even the older versions use reflection and IL code generation to optimize getting properties from entities. However, this is *only* the case for `RealPropertyExport` and `ComplexPropertyExport`. Both `***VoltageExport` and `***CurrentExport` will not use reflection and work on AoTC platforms. The `***CurrentExport` will only work on behaviors that define a dedicated current branch in the system of equations, which is the case for the behaviors of a `VoltageSource` (and dependent voltage sources), or an `Inductor`. Resistors don't define their own current branch so that will not work.

Instead, consider placing a (perhaps hidden) 0V voltage source in series and get the current from that using `RealCurrentExport`.

Alternatively, you can usually avoid using exporters altogether, and just use reflection every loop. It will be slower, but I believe it will work on AoT platforms.

I have decided to solve this problem in the future using Code Generators (see issue #99 ). I expect this to be a rather larger change so it will take some time before it takes effect.

svenboulanger, 2021-02-19:
Just curious, do the `.SetParameter<P>("x", value)` and `.GetProperty<P>("y")` methods work for you? They should be using reflection, but I know that a lot of AOT compilers support a small set of methods so I was wondering these methods need change.

BlackBerry007, 2021-02-19:
unfortunately it does not work.

svenboulanger, 2021-02-19:
Ok thanks for checking! I am considering removing reflection altogether in favor of source generators. This would make it compatible with any platform.

Maybe one final question, which framework are you targeting? In order to use source generators, I at minimum need .NET Standard 2.0 (Spice# 2x/3x currently support .NET Standard 1.6 as well, so I would have to drop support for that).

BlackBerry007, 2021-02-19:
I am developing in Unity and the api compatibility level there is .NET Standard 2.0 or .NET 4.x or newer. So that would be no problem I guess.

svenboulanger, 2021-04-25:
It should be fixed starting from Spice# v3.1.0 which doesn't use reflection anymore (except purely for documentation purposes).

closed 2021-04-25 by svenboulanger