SpiceSharp/SpiceSharp #140: Question: Watt/Power Meter Implementation

Joshiii, 2021-02-17:
Sorry for asking this question, I know this may be a basic/simple question for someone who can read this but I'm literally new in using SPICE and have found some fresh air when I found SpiceSharp. Given the case, I'm still having trouble on the implementation on my program. To have you some background, I'm working on a project where the Canvas contains fixed number (10) of resistor, inductor and capacitor. The program should be able has Ammeter, Voltmeter and Wattmeter. I think I'm good with the Ammeter and Voltmeter. This time, I need to find the power between 2 pins (for wattmeter). This is what I found so far.

[image]

Can you help me translate it to SpiceSharp? Thanks!

svenboulanger, 2021-02-17:
I'm not sure where the issue is. You can actually get the power dissipated by the resistor directly by asking the resistor (ask it for the "p" property).

```csharp
var ckt = new Circuit(
    // ...
    new Resistor("R1", "2", "0", 1e3));

var op = new OP("op");
op.ExportSimulationData += (sender, args) => {
    var power = op.EntityBehaviors["R1"].GetProperty<double>("p");
};
op.Run(ckt);
```

In the more general case, you would need to multiple the voltage across the two nodes with the current from node 1 to node 2. I suspect that is why the VSENSE1 source was added, as it is a typical Spice-trick to measure the current through branches.

If you have more general questions about electronics, I suggest going to [StackExchange](https://electronics.stackexchange.com/). If it is still strongly Spice#-related, you can also try using the Discussions-tab on this GitHub page which kind of seems to emulate the StackExchange concept.

Joshiii, 2021-02-18:
Thank you for your response. Highly appreciated.

Yes, I can get the power dissipated by the resistor using that code, problem is the parameter available when I'm about to get the power is Node 1 and 2. So I don't have the specific resistor to read. See below for what I'm trying to simulate. (This is just a mock up):

[image]

So I'm trying to do the way SPICE gets current using 2 nodes (By putting VoltageSource with 0V). That's why I'm trying to refer to the SPICE Power Meter image in my post.

P.S.: Sorry if I directly posted this question on issue page, I've been browsing StackOverFlow of questions about SpiceSharp and it seems most of the author is not receiving enough response.

svenboulanger, 2021-02-18:
You cannot make a Watt-meter with 2 connections (like in your figure). You will always need a current and a voltage probe, and then multiply those measurements.

It should be pretty straight-forward:
- You measure the voltage between node 1 and 2
- You measure the current using that voltage source of 0V
- You multiply the voltage and the current in code to get the power.

Joshiii, 2021-02-20:
Got it. Wattmeter is now working. Thank you for your time. SpiceSharp is a great help for my project.

closed 2021-04-25 by svenboulanger