SpiceSharp/SpiceSharp #160: Diode model issue

marcin-golebiowski, 2021-05-02:
It seems that model for diode has some issue with positive breakdown voltage.

E.g. Let's take the values from https://www.el-component.com/diodes/1n4148

.model 1N4148 D 
+ IS = 4.352E-9 
+ N = 1.906 
+ BV = 110 
+ IBV = 0.0001 
+ RS = 0.6458 
+ CJO = 7.048E-13 
+ VJ = 0.869 
+ M = 0.03 
+ FC = 0.5 
+ TT = 3.48E-9

Spice# is throwing exception:

System.ArgumentOutOfRangeException: 'The parameter value should be less than 0.
Parameter name: BreakdownVoltage
Actual value was 110.'

marcin-golebiowski, 2021-05-02:
I see that there is already something about it #141
Is it possible to extend model to accept positive breakdown voltage ?

svenboulanger, 2021-05-02:
Strictly speaking, Spice# is correct if you follow the conventions for any diode IV curve you'll find. Is this to preserve consistency with other Spice simulators?

marcin-golebiowski, 2021-05-03:
LTSpice and TINA support positive BV. There are some examples of netlist where BV is positive.
I think Spice# should support to be compatible with other simulators.

svenboulanger, 2021-05-05:
Will add this to the models with the next update.

svenboulanger, 2021-05-08:
The code actually assumed a positive value, and the limit should've been positive.
But in any case, both will be accepted by Spice#.