SpiceSharp/SpiceSharp #171: Any way to create wire?

AzadKshitij, 2021-06-29:
Is there any way to create wire or something similar to that?

svenboulanger, 2021-06-29:
Hi,

If you are looking for something to short-circuit two nodes together, the usually recommended way would be to use a 0V voltage source. If the wire is ohmic, you can just use the `Resistor`-component.

If not: a wire is used in schematics to denote everything that is at the same potential, i.e. a single node in Spice#. Spice# does not have any concept of a (graphical) wire and neither does any other Spice simulator. It is only something you will see/use in *schematic editors*, which will translate everything connected by wires to nodes before sending it to the simulator.

Yours sincerely,
Sven

AzadKshitij, 2021-06-29:
Thank you that Worked....

closed 2021-06-29 by svenboulanger