SpiceSharp/SpiceSharp #172: Astable Multivibrator

LarsSem, 2021-07-07:
Hi, 

currently i am trying to simulate an astable multivibrator with spciesharp. To get it into one state at the beginning, I used different preconditions. 
But in the transient simulation it gets stuck in this state and I get a TimeStepTooSmallException.
I have already tried different initial conditions, component values and model parameters without success.
I am not sure if it is a problem with SpiceSharp or with my circuit.
I hope someone here can help me.

        var bjtModelNPN = new BipolarJunctionTransistorModel("bjtModel");
        bjtModelNPN.SetParameter("npn", true);
                    
        var ckt = new Circuit(
            new VoltageSource("V1", "a", "0", 12),
            new Resistor("R1", "a", "b", 12000),
            new Resistor("R2", "a", "c", 300000),
            new Resistor("R3", "a", "d", 300000),
            new Resistor("R4", "a", "e", 12000),
            new Capacitor("C1", "b", "c", 10e-9),
            new Capacitor("C2", "e", "d", 10e-9),
            new BipolarJunctionTransistor("T1","b","d","0","0",bjtModelNPN.Name),
            new BipolarJunctionTransistor("T2", "e", "c", "0", "0", bjtModelNPN.Name),
            bjtModelNPN            
            ) ;

        var trans = new Transient("trans", 0.0001, 0.02);
                
        trans.TimeParameters.InitialConditions["b"] = 12;
        trans.TimeParameters.InitialConditions["c"] = 0;
        trans.TimeParameters.InitialConditions["d"] = 1;
        trans.TimeParameters.InitialConditions["e"] = 0;

svenboulanger, 2021-07-08:
I think it is a model error. I don't think the default BJT was supposed to be used without specifying any parameters.

I plugged this in and the simulation worked fine for me (I took it from Marcin's Spice#.Parser integration tests):

```csharp
bjtModelNPN.SetParameter("npn", true)
    .SetParameter("bf", 200.0)
    .SetParameter("cjc", 20e-12)
    .SetParameter("cje", 20e-12)
    .SetParameter("is", 1e-16);
```

LarsSem, 2021-07-08:
Thank you
I didn't realize that some parameters have no default value and I looked for the error in all other places.

closed 2021-07-09 by LarsSem