SpiceSharp/SpiceSharp #183: Time to get Transient simulation

Angelo50, 2021-09-24:
Hi Sven,

May be you ore someone else could enlight me.
I supposed it is more a missunderstood of myself than an issue.

here is the problem:

I have buid a circuit with 5 Transitors (2N2222 models), 5 Leds and 6 capacitors, just as an example of blinking Leds.
My problem is until 4 NPN transistor, the time response of simulation is immediate but starting 5, it takes around 1 minute.

here is the corresponding netlist circuit: (generated by my c#)
The V nodes with 0 Volt are just connections

* New Generated Circuit
V1000 1 0 6
V0 1 2 0
V1 2 3 0
V2 3 4 0
R3 3 5 470
.model LedSimulation D (IS=93.2e-11 RS=42m N=3.73)
D4 5 6 LedSimulation
R5 4 7 32000
.model bjtModel NPN (is=2.2e-15 BF=240 CJE=24.9e-12 vje=1.10 cjc=12.4e-12 vjc=0.300 mjc=0.300 tf=371e-12 tr=64e-9 eg=1.12)
Q7 10 9 8 0 bjtModel
V8 6 10 0
V9 8 11 0
V11 12 0 0
V13 11 14 0
V14 14 12 0
C15 13 14 2,2E-06
V16 6 15 0
V17 15 13 0
V18 15 16 0
V19 16 17 0
C20 17 18 2,2E-06
V21 7 9 0
V23 4 19 0
V24 19 20 0
R25 19 21 470
D26 21 22 LedSimulation
R27 20 23 32000
Q29 26 25 24 0 bjtModel
V30 22 26 0
V31 24 27 0
V32 27 11 0
V33 23 25 0
C35 7 22 2,2E-06
V36 20 28 0
V37 28 29 0
R38 28 30 470
R39 29 31 32000
D40 30 32 LedSimulation
Q42 35 34 33 0 bjtModel
V43 32 35 0
V44 33 36 0
V45 36 27 0
V46 18 37 0
V48 31 34 0
C49 23 32 2,2E-06
V50 29 38 0
V51 38 39 0
R52 38 40 470
R53 39 41 32000
D54 40 42 LedSimulation
C55 42 31 2,2E-06
Q56 45 44 43 0 bjtModel
V57 42 45 0
V58 43 46 0
V59 46 36 0
V60 41 44 0
V62 47 37 0
V63 39 48 0
V64 48 49 0
R65 48 50 470
D66 50 51 LedSimulation
R67 49 52 32000
C68 51 41 2,2E-06
Q69 55 54 53 0 bjtModel
V70 52 54 0
V71 51 55 0
V72 53 56 0
V73 56 46 0
V75 57 47 0
V76 54 57 0

here is the simulation c#launch 👍 

 var tran = new Transient("dc", 0, 100, 0.0001);
 tran.TimeParameters.UseIc = true;
 tran.TimeParameters.TransientMaxIterations = 10;
 tran.TimeParameters.StartTime = 0.1f;

Thanks for your work.

Angelo.

svenboulanger, 2021-09-24:
It is certainly possible that the time of a simulation varies significantly. Spice is a variable timestep simulator, so if you have harshly nonlinear things going on then the time it takes to complete a simulation will take longer because it computes more timesteps.

If you want more consistent simulation times, consider going for Euler method, which has a fixed timestep. The obvious drawback is that the error will be higher, but perhaps that is not an issue for your use-case.

Angelo50, 2021-09-29:
Again, thanks for your answer
May I ask you to give me a link of an exemple for euler method ?
I'm very sorry but I definitely have problem to find documentation on SpiceSharp
the only one I have is the one attached of the downloaded nuggets and it lacks of example  (I need example to understand)

Angelo.

svenboulanger, 2021-10-01:
There is an example in the [test project](https://github.com/SpiceSharp/SpiceSharp/blob/dace7bddaadf7ac22bc32abd2dc61b4d28afb13d/SpiceSharpTest/Simulations/TransientTests.cs#L115).

Angelo50, 2021-10-05:
Thanks Sven.
I have just discovered this testProject with your help, and it is so helpfull for me

Angelo.