SpiceSharp/SpiceSharp #200: Problems when simulating transformer

barna-leopoly, 2022-03-30:
Hi,

I'm currently using Spice# in a project where the user can create simple circuits from pre-made components (3D environment, Unity). Things are working great, except for circuits with a transformer in them.

Couple of things to note: 

- I use "zero" ohm resistors to represent cables.
- Due to the nature of the project I found FixedEuler to be the best way to export the necessary values.
- I use a "dummy" resistor to connect the two sides of the transformer circuit.

I found several issues during my testing. The first is when the voltage source (sine wave) is placed on the secondary inductor's side, things work in weird ways or don't work at all. The following code creates a circuit where trying to run the simulation throws an exception with the text "Could not determine operating point." Perhaps this is the intended behaviour, but it is definitely worth mentioning.

<details><summary>Voltage source at secondary inductor</summary>
<p>

```csharp
    /* o: nodes, r: resistor, ~: AC source, {}: transformer coils, -|: cables, L: lamp (resistor) m: mutual inductance
     * o r o r o - o
     * ~   {   }   |
     * o - o m o L o
     */
    void TransformerACAtSecondary()
    {
        var cable1 = new Resistor("cable1", "0", "1", 0.0);
        var mi = new MutualInductance("mi", "coil2", "coil1", 1);
        var lamp = new Resistor("lamp", "2", "3", 10.0);

        var resistor1 = new Resistor("resistor1", "4", "5", 1.0);
        var dummy = new Resistor("dummy", "5", "6", 1e12);
        var cable2 = new Resistor("cable2", "6", "7", 0.0);

        var ac = new VoltageSource("ac", "0", "4", new Sine(0.0, 9.0, 1.0));
        var coil1 = new Inductor("coil1", "1", "5", 25);
        var coil2 = new Inductor("coil2", "2", "6", 25);
        var cable3 = new Resistor("cable3", "3", "7", 0.0);

        var ckt = new Circuit(
            cable1,
            mi,
            lamp,
            resistor1,
            dummy,
            cable2,
            ac,
            coil1,
            coil2,
            cable3
            );

        var tran = new Transient("Tran 1", new FixedEuler { Step = 0.02, StopTime = double.PositiveInfinity });
        current0Export = new RealPropertyExport(tran, "lamp", "i");
        current1Export = new RealPropertyExport(tran, "cable1", "i");

        tran.ExportSimulationData += (sender, args) =>
        {
            Debug.Log("Lamp: " + current0Export.Value.ToString() + " at: " + args.Time.ToString());
            Debug.Log("Cab1: " + current1Export.Value.ToString() + " at: " + args.Time.ToString());
            double length = 1.0;
            if (args.Time > (length * 10.0 - 0.0001))
            {
                tran.TimeParameters.StopTime = 0.0;
            }
        };

        tran.Run(ckt);
    }
```

</p>
</details>

Even when the voltage source is on the "correct" side problems arise when some parts of the circuit aren't connected e.g. there's a cable that only has a connection on one end. This can often happen in my case because users create the circuit piece by piece.

The following example creates a circuit where the voltage source is connected to the primary inductor, while the secondary inductor's side does not create a closed circuit. Yet the resistor named "lamp" receives very high current.

<details><summary>Non-closed circuit at secondary inductor</summary>
<p>

```csharp
    /* o: nodes, r: resistor, ~: AC source, {}: transformer coils, -|: cables, L: lamp (resistor) m: mutual inductance
     *                 o
     *                 |
     * o r o r o - o L o
     * ~   {   }
     * o - o m o - o
     */
    void TransformerIncomplete()
    {
        var cable1 = new Resistor("cable1", "0", "1", 0.0);
        var mi = new MutualInductance("mi", "coil1", "coil2", 1);
        var cable2 = new Resistor("cable2", "2", "3", 0.0);

        var resistor1 = new Resistor("resistor1", "4", "5", 1.0);
        var dummy = new Resistor("dummy", "5", "6", 1e12);
        var cable3 = new Resistor("cable3", "6", "7", 0.0);
        var lamp = new Resistor("lamp", "7", "8", 10.0);

        var ac = new VoltageSource("ac", "0", "4", new Sine(0.0, 9.0, 1.0));
        var coil1 = new Inductor("coil1", "1", "5", 25);
        var coil2 = new Inductor("coil2", "2", "6", 25);

        var cable4 = new Resistor("cable4", "8", "9", 0.0);

        var ckt = new Circuit(
            cable1,
            mi,
            cable2,
            resistor1,
            dummy,
            cable3,
            lamp,
            ac,
            coil1,
            coil2,
            cable4
            );

        var tran = new Transient("Tran 1", new FixedEuler { Step = 0.02, StopTime = double.PositiveInfinity });
        current0Export = new RealPropertyExport(tran, "lamp", "i");
        current1Export = new RealPropertyExport(tran, "cable3", "i");

        tran.ExportSimulationData += (sender, args) =>
        {
            Debug.Log("Lamp: " + current0Export.Value.ToString() + " at: " + args.Time.ToString());
            Debug.Log("Cab3: " + current1Export.Value.ToString() + " at: " + args.Time.ToString());
            double length = 1.0;
            if (args.Time > (length * 10.0 - 0.0001))
            {
                tran.TimeParameters.StopTime = 0.0;
            }
        };

        tran.Run(ckt);
    }
```

</p>
</details>

My final example shows a case where there are closed circuits on both sides, but there's also a "dangling" row of cables on the secondary side. This also causes a "Could not determine operating point" exception.

<details><summary>Dangling cable at secondary inductor</summary>
<p>

```csharp
    /* o: nodes, r: resistor, ~: AC source, {}: transformer coils, -|: cables, L: lamp (resistor) m: mutual inductance
     *         o - o - o
     *         |       |
     * o r o m o - o   o
     * ~   {   }   L   |
     * o - o r o - o   o
     *                 |
     *             o - o
     */
    void TransformerDanglingWire()
    {
        var cable0 = new Resistor("cable0", "1", "0", 0.0);

        var cable1 = new Resistor("cable1", "2", "3", 0.0);
        var dummy = new Resistor("dummy", "3", "4", 1e12);
        var cable2 = new Resistor("cable2", "4", "5", 0.0);

        var resistor1 = new Resistor("resistor1", "6", "7", 1.0);
        var mi = new MutualInductance("mi", "coil1", "coil2", 1);
        var cable3 = new Resistor("cable3", "8", "9", 0.0);

        var cable4 = new Resistor("cable4", "10", "11", 0.0);
        var cable5 = new Resistor("cable5", "11", "12", 0.0);

        var cable6 = new Resistor("cable6", "1", "13", 0.0);

        var ac = new VoltageSource("ac", "6", "2", new Sine(0.0, 9.0, 1.0));
        var coil1 = new Inductor("coil1", "3", "7", 25);
        var coil2 = new Inductor("coil2", "4", "8", 25);
        var lamp = new Resistor("lamp", "5", "9", 10.0);
        var cable7 = new Resistor("cable7", "13", "14", 0.0);

        var cable8 = new Resistor("cable8", "8", "10", 0.0);
        var cable9 = new Resistor("cable9", "14", "12", 0.0);

        var ckt = new Circuit(
            cable0,
            cable1,
            dummy,
            cable2,
            resistor1,
            mi,
            cable3,
            cable4,
            cable5,
            cable6,
            ac,
            coil1,
            coil2,
            lamp,
            cable7,
            cable8,
            cable9
            );

        var tran = new Transient("Tran 1", new FixedEuler { Step = 0.02, StopTime = double.PositiveInfinity });
        current0Export = new RealPropertyExport(tran, "lamp", "i");
        current1Export = new RealPropertyExport(tran, "cable2", "i");
        current2Export = new RealPropertyExport(tran, "cable3", "i");

        tran.ExportSimulationData += (sender, args) =>
        {
            Debug.Log("Lamp: " + current0Export.Value.ToString() + " at: " + args.Time.ToString());
            Debug.Log("Cab2: " + current1Export.Value.ToString() + " at: " + args.Time.ToString());
            Debug.Log("Cab3: " + current2Export.Value.ToString() + " at: " + args.Time.ToString());
            double length = 1.0;
            if (args.Time > (length * 10.0 - 0.0001))
            {
                tran.TimeParameters.StopTime = 0.0;
            }
        };

        tran.Run(ckt);
    }
```

</p>
</details>

I hope we can find some solution to these problems.
Thanks for any help and for creating an otherwise excellent library.

svenboulanger, 2022-04-02:
Hi, glad to see you using Spice#!

That does seem a bit strange to me. I'll investigate it further.

svenboulanger, 2022-04-02:
I was looking at the solver matrix for the first example, and noticed that the solver could not solve it due to floating point arithmetic limitations.

It is actually not possible for Spice# to simulate 0ohm resistors, it internally limits them to 1e-12ohm. This, combined with the very large resistor of 1e12 ohm leads to a very badly conditioned matrix. In general, the bigger the difference is between the largest and smallest value amplitudes, the more difficult it can be for the solver to solve the matrix.

Can you please re-try your examples with maybe more limited numbers? E.g. the open at around 1e9ohm, and the resistances around 1e-3 ohm.

barna-leopoly, 2022-04-04:
I tried your suggestion first with 1e9 and 1e-3 ohm resistors, later with 1e6 and 1e-6 ohms, and both seem to work without issue.

Thanks for the answer, you helped me a great deal!

svenboulanger, 2022-04-04:
No problem. Glad to hear it's resolved! If I remember correctly, other Spice simulators will not use 1e-12Ohm resistor but a 1mOhm resistor (e.g. like in [ngSpice]( https://github.com/ngspice/ngspice/blob/032b1c32c4dbad45ff132bcfac1dbecadbd8abb0/src/spicelib/devices/res/restemp.c#L74)). Maybe I should follow suit to be safe...

If you want to have a strict short, use a 0V voltage source instead (this is also the advise since the original Spice simulator). I'll close this issue :-)

closed 2022-04-04 by svenboulanger