SpiceSharp/SpiceSharp #206: Pausing and Resuming simulation

OmegaHaxors, 2022-09-06:
I'm planning on using SPICE# as an engine for a game mod. Right now I have some code written to run the simulator for 1 time step (1 second) then save the values within the inductors and capacitors to the game's treeattribute system and then when the next tick begins (which will happen once every second) load those values and start the simulator again.

However when starting a simulator with a VoltageSource, it will start the simulator with the voltage having already propagated across the entire network, as if an infinite amount of time was allowed to pass before it started. This means that as soon as the transient analysis begins, any value stored within capacitors is completely clobbered by the value coming from any connected VoltageSources.

OmegaHaxors, 2022-09-06:
[image]

OmegaHaxors, 2022-09-07:
Reopening because another unrelated issue has cropped up.

This is a unique problem due to the circumstances of the environment:
InitialConditions allows me to set voltage/current at points based on the name of the pin/node respectively. This works.
However, due to the entire state of the circuit being converted into strings and values in an Named Binary Tree, I can't rely on VoltageSource.Parameters.DcValue like I did in my old test code. I need to somehow get and set the node's DcValue parameter using only the node's name alone. The API has been less-than-helpful in finding an answer and I doubt anyone else has had to deal with such an unusual problem. Think you could give me a hand with this weird situation?

svenboulanger, 2022-09-07:
What you are trying to describe (I think) is the "second" form of initial conditions which are specified on the components themselves. For example, a capacitor has an "ic" parameter that specified the initial voltage across it. Same for an inductor that specifies the initial current through it.

The problem is that if your initial conditions are too rough or unmatched, the simulator will experience a severe glitch and might not even converge. Maybe it is worth a try?

OmegaHaxors, 2022-09-07:
I'll let you know how it turns out but I'm going to need to know how to get the components from tran. or ckt. first

closed 2022-09-07 by OmegaHaxors

OmegaHaxors, 2022-09-07:
Hey just getting back to you. Did
ckt.TryGetEntity(capacitornodename, out var capacitornode);
capacitornode.SetParameter("ic",capacitorvoltage);
and it worked like a dream. Thanks a ton for help.