SpiceSharp/SpiceSharp #207: Not sure why this is failing to validate

OmegaHaxors, 2022-09-26:
I've been making a lot of great progress on implementing SpiceSharp into a game through a mod, but after overhauling the internal code all of a sudden it's causing issues. To allow players to build circuits out in real time I added in some special handling to top off any floating nodes with parasitic capacitors since transients will crash if there are any floating nodes.

For some reason, the game crashes with 1 violation whenever I place a new node object in game. What am I doing wrong here?

[image]

svenboulanger, 2022-09-27:
The transient is probably fine, but in DC you still have floating nodes. The way that traditional Spice simulators solve this issue, is that they add a very large resistance to ground instead of a capacitor (e.g. 1e12 ohms).

A normal transient simulation will first compute a DC solution, assuming that capacitors are open circuits, hence the rule violation.

OmegaHaxors, 2022-09-27:
Wow I never would have figured that one out on my own. Is it possible to disable the DC analysis?

OmegaHaxors, 2022-09-29:
Going to close since the issue has been fixed.

closed 2022-09-29 by OmegaHaxors

svenboulanger, 2022-09-29:
There is a way to bias the DC simulation part, and that is by using the `UseIc` parameter (in Spice, this is the same as the UIC flag on transient simulations).

You'll have to specify initial values for all nodes yourself, give all devices an initial state and if those values are not close enough to something that solves, the simulator might not converge.
So in the end, I haven't seen UIC being used all that often. It's just much easier to have the simulator solve it for you.