SpiceSharp/SpiceSharp #219: PropertySet is empty for Inductors

das-codes03, 2024-01-15:
Hello, first of all this is a nicely written and well documented project.

Coming to my issue:

I am creating a 3D based simulation solution in unity. While playing around with various components, I faced the following issue:

While trying to to modify resistance value and capacitance value during transient simulation in runtime, it works perfectly. However, incase of **inductors**, the PropertySet is empty. 

I went through the source code but couldn't find anything wrong. Maybe I am missing something while compiling from source, or is there a bug, or do inductors not support this feature?

        // Create a circuit
        Transient sim;
        var ckt = new Circuit(
            new VoltageSource("V1", "0", "1", 1.0),
            new Inductor("L1", "1", "2", 1.0),
            new Resistor("R1", "2", "0", 1.0)

        );

        // Create a transient simulation
        sim = new Transient("Tran", 0.0001, 100000, 0.0001);
        var export = new RealPropertyExport(sim, "V1", "i");

        //*Issue*//
        SpiceSharp.Components.Inductors.Parameters bp = null;
        ITemperatureBehavior tb = null;
        sim.AfterSetup += (sender, args) =>
        {
            BehaviorContainer eb = (BehaviorContainer)sim.EntityBehaviors["L1"];
            tb = eb.GetValue<ITemperatureBehavior>();

            //this is empty in case of inductors i.e false
            Debug.Log(eb.TryGetParameterSet(out bp));
            //************************************
            //works fine incase of resistors and capacitors

        };
        // Register to the event for exporting simulation data
        sim.ExportSimulationData += (sender, exportDataEventArgs) =>
            {
                if (!alive)
                {
                    thread.Abort();
                }
                var time = exportDataEventArgs.Time;
                var currentThroughR1 = export.Value;
                // Debug.Log($"Time: {time} {t}");
                while (t < time)
                {
                    //wait
                }
                currents.Enqueue(new pt(time, currentThroughR1));
            };
        sim.BeforeLoad += (s, e) =>
        {
            //here the parameter will be modified
            tb.Temperature();
        };

        // Run the simulation
        sim.Run(ckt);`

svenboulanger, 2024-01-17:
If I remember correctly, an inductor doesn't have a temperature behavior like a resistor or a capacitor. Can you try using the parameter set instead for updating the inductance?

das-codes03, 2024-01-17:
I got it to work finally.
This is the code section to update inductance.

`BehaviorContainer eb = (BehaviorContainer)sim.EntityBehaviors["L1"];
tb = eb.GetValue<ITemperatureBehavior>();
((Inductor)ckt["L1"]).Parameters.Inductance = 0.1;
tb.Temperature();`

Actually the problem i was getting previously is that when doing `eb.ParameterSets.Count()` incase of inductors returns 0. I directly updated the inductance and it works. But yes, temperature had to be called to actually have any effect.

I am not sure why that was the case (ParameterSets being empty), is it an expected behavior?

Thank you

svenboulanger, 2024-01-17:
I'm not sure, but it doesn't sound correct. I'll take a look if I find the time time.

svenboulanger, 2024-01-31:
Good job finding this. The inductor behavior was missing an `IParameterized<Parameters>` interface, which allows `TryGetParameterSet<...>` to work. I will go through all the behaviors again and see if I missed any others.

commit fdc73a50 (2024-01-31): Fix #219 - missing interface for finding param set
closed 2024-01-31 by svenboulanger