SpiceSharp/SpiceSharp #237: How to implement a speaker as a Custom Components

vudzero, 2024-08-17:
Hi!

I want to simulate a circuit with a speaker, my understanding is that I would need to create a custom component. In the audio world, speakers are described with what is called a .zma file which provides the impedance and phase for a given list of frequencies. It is usually a list of around 400 measurements between 20Hz and 20kHz

Example:
**Freq,Z,Phase**
20,40,-12
40,80,-18
etc

What would be the easiest way to implement a custom element which would rely on this list of known values?

Thanks!

vudzero, 2024-08-17:
If that can helps, this is the way to do it un LTSpice

.subckt MAXI330 1 2 
B1 1 2 V=I(B1)  MAG FREQ=
+ 20.142, 20.883, -74.570
+ 50.537, 8.537, -52.902
+ 98.145, 6.676, 4.844
+ 494.019, 5.673, 19.204
+ 1017.700, 9.171, 39.194
+ 4983.032, 21.734, -10.051
+ 9965.698, 8.458, 25.251
+ 45714.109, 42.504, 65.181
.ends

svenboulanger, 2024-08-19:
Does this include transient simulations? For AC-only simulations it should be fairly easy to implement, and you will only need some interpolation algorithm to find mag/phase values in between frequency points.

If you want to do this like LTSpice does it in transient simulation, then IIRC you will need to do an inverse fourier transform and turn the spectrum into an impulse response. This impulse response can then be convolved with whatever input signal enters the speaker. Not trivial but possible with time and effort.